tshakalekholoane/bat #1: exit status 1

sjv-afk1, 2021-02-06:
I followed the instructions, but it doesn't seem to work. My laptop is Asus tuf A17  (fa706ih-au054t). I am getting this following message : "exit status 1". This is displayed for any command I run.

leveson, 2021-02-06:
Would you mind providing more information with regards to that?

For starters, was that the only message you got? Did you download the application from [the releases page](https://github.com/leveson/bat/releases/download/0.2/bat) or did you build it from source? Which Linux distribution are you using?

sjv-afk1, 2021-02-06:
> I suspect it might have something to do with permissions.
> 
> Please let me know if running the following command solves the issue.
> 
> ```shell
> chmod +x $HOME/Downloads/bat
> ```
> 
> Assuming the application is in your `Downloads` folder.

So, I am assuming, you mean to say by using chmod command, you want me to make it executable. This I already did with using GUI. Anyway I tried the using the shell also and this is the output I am getting-

//

x@x~/Downloads$ chmod +x $HOME/Downloads/bat
x@x~/Downloads$ ./bat --threshold
2021/02/06 23:29:05 exit status 1
x@x:~/Downloads$ 

//

leveson, 2021-02-06:
Thank you for pointing this out. I will try to run it on Ubuntu and see if I can pin it down.

leveson, 2021-02-06:
> > I suspect it might have something to do with permissions.
> > Please let me know if running the following command solves the issue.
> > ```shell
> > chmod +x $HOME/Downloads/bat
> > ```
> > 
> > 
> > Assuming the application is in your `Downloads` folder.
> 
> So, I am assuming, you mean to say by using chmod command, you want me to make it executable. This I already did with using GUI. Anyway I tried the using the shell also and this is the output I am getting-
> 
> //
> 
> x@x~/Downloads$ chmod +x $HOME/Downloads/bat
> x@x~/Downloads$ ./bat --threshold
> 2021/02/06 23:29:05 exit status 1
> x@x:~/Downloads$
> 
> //
> 
> I am using ubuntu 20.04 by the way.

Would you mind showing me the output of the following commands?

```shell
uname -s
```
and

```shell
/sys/class/power_supply/BAT?/
```

I can confirm that I am getting the same error but I am running Ubuntu 20.04 on a virtual machine which has some limitations.

sjv-afk1, 2021-02-06:
> ```shell
> ls -A /sys/class/power_supply/BAT?/
> ```

x@x:~$ uname -r
5.8.0-41-generic

x@x:~$ ls -A /sys/class/power_supply/BAT?/
alarm                         charge_now    model_name     technology
capacity                      current_now   power          type
capacity_level                cycle_count   present        uevent
charge_control_end_threshold  device        serial_number  voltage_min_design
charge_full                   hwmon2        status         voltage_now
charge_full_design            manufacturer  subsystem

leveson, 2021-02-07:
Thank you. Working on a fix.

leveson, 2021-02-07:
Please let me know if [this](https://github.com/leveson/bat/files/5938170/bat.zip) fixes the issue.

leveson, 2021-02-07:
Please let me know if [this](https://github.com/leveson/bat/files/5938197/bat.zip) fixes the issue.

sjv-afk1, 2021-02-07:
Yes, it seems that it is working. If I face any issues, I will let you know. Will do thorough testing. Thank you so much!
Is there any interesting project on Github or Gitlab which covers power saving techniques, like there is TLP for lenovo laptops? I tried various forums and could not find any worthy solution. The experience of running a gnu/linux distibution on battery mode has been disappointing so far :(

leveson, 2021-02-07:
Cheers. Thank you for your help with the issue.
No, I do not know of any projects with regards to that.

closed 2021-02-07 by leveson