tshakalekholoane/bat #2: this looks pretty cool, not really an issue

nyck33, 2021-02-09:
Hi Leveson,

  I just bought an Ideapad to replace my T420 Thinkpad because I accidentally cracked the screen.  I saw your answer on askubuntu I think and clicked to get here.  
  Did you just recently make this?  I'm surprised you don't have more stars.  
Anyways, if you have a moment, please feel free to contact me.

Thanks and regards,

Nobu

leveson, 2021-02-09:
I am not certain. Perhaps you could give it a try?

My gut feeling is that how the underlying hooks are exposed may well be vendor specific which would mean in its current state it only works for ASUS laptops. But it is possible that this is not the case or if so that the application could be extended to provide the same functionality for Levono laptops with some minor modifications.

nyck33, 2021-02-10:
Thanks for responding.
I actually found out that on Lenovo Ideapads, if I set the Lenovo battery conservation utility in Windows 10, it seems to set it for the hardware regardless of the OS I boot into.
For example, I can boot into Ubuntu on an external SSD after pressing F12 before boot to go into Bios, select the USB device and Ubuntu shows that the charge never goes above a certain level.
Lenovo says 60% for Windows 10 but it looks a bit higher on Ubuntu.  Mind you this SSD with Ubuntu used to be internal on a Thinkpad T420 and I did all the stuff with TLP to set charge 30 80.
But this magic did not happen until I went into Windows 10 and set the Lenovo battery conservation.

leveson, 2021-02-10:
Well, by my understanding, these sorts of applications work by manipulating kernel settings which generally do not persist between restarts. Instead, these settings are applied each time the system boots by some user space tool.

I do not know how this is done in Windows but in Linux this achieved through manipulating virtual files in the [`/sys/`](https://man7.org/linux/man-pages/man5/sysfs.5.html) directory and to achieve persistence one would set some daemon like [sytemd](https://systemd.io) to listen for events such when the system boots and then call on the application to reapply the setting.

My guess is that what is happening in your case is not that Windows is setting the threshold universally so to speak but when you boot into Windows, the Lenovo utility kicks in and when you boot into Ubuntu your old TLP settings do the same which would explain the discrepancy between the charging threshold between the different operating systems.

leveson, 2021-02-12:
Going to close the issue. You can send me an email if you want to discuss this or anything further.

closed 2021-02-12 by leveson

nyck33, 2021-02-16:
That's awesome Eric, thanks for your insight.

Regards,

Nobu