tshakalekholoane/bat #4: Authentication failure

fabricio8800, 2021-03-11:
When I execute the command below it returns "Authentication failure.":

./bat --threshold 60

The problem happens using root or sudo.

leveson, 2021-03-11:
Which distribution are you running?

fabricio8800, 2021-03-11:
Fedora 33.

leveson, 2021-03-11:
I am working on a fix to a related problem that appears on Ubuntu. When that is done, I will post the binary and then you can let me know if it fixes your problem as well? If not then perhaps we can explore it further.

leveson, 2021-03-11:
Can you let me know if this fix works before I put it to release?

[bat.zip](https://github.com/leveson/bat/files/6122818/bat.zip)

fabricio8800, 2021-03-11:
New error:

[root@fedora Downloads]# ./bat --threshold 80
panic: runtime error: index out of range [0] with length 0

goroutine 1 [running]:
main.setThreshold(0x50)
	/home/lvsnxo/dev/bat/bat.go:165 +0x40c
main.main()
	/home/lvsnxo/dev/bat/bat.go:219 +0x3fc

leveson, 2021-03-11:
What's the manufacturer/model of your laptop?

fabricio8800, 2021-03-11:
Is a Dell Inspiron 5368.

leveson, 2021-03-11:
I am afraid I don't think this will work with your laptop. 

The variables that control the charging threshold are vendor specific, they even vary within different models from the same manufacturer and so far this has only been tested on ASUS machines.

I will research the problem, but your best bet might be [this command-line application from Dell](https://www.dell.com/support/kbdoc/en-uk/000178000/dell-command-configure) that can do this sort of thing and some scripts from [this repository](https://github.com/glynhudson/dell-charge-limit) that abstract parts of the application to set the charging threshold.

fabricio8800, 2021-03-11:
Ok, thanks.

leveson, 2021-03-11:
Cheers ;)

closed 2021-03-11 by leveson

fabricio8800, 2021-03-12:
I solve my problem with:

smbios-battery-ctl --set-custom-charge-interval "low high" and smbios-battery-ctl --set-charging-mode custom

leveson, 2021-03-12:
That's interesting. Did you have to install anything or did that come with your system?

fabricio8800, 2021-03-12:
I installed the smbios-utils package.

leveson, 2021-03-12:
Thank you for letting me know. I think I will update the readme to redirect anyone with a Dell laptop to that package.