tshakalekholoane/bat #5: Setting is not persisting after reboot

lemurweb-dev, 2021-03-25:
Hello! First of all, nice tiny package!
I want to issue something:

#### 1
When i run `bat -t 70`, it asks for my root password but authentication fails every time.
If i then run `sudo bat -t 70`, no issue and setting is correctly set.

#### 2
I run `sudo bat -p` multiple times but the threshold value at every restart is back at 100.
What can i do to make it persistent?

I'm using elementaryOS:
```
catta@catta-pc:~$ uname -a
Linux catta-pc 5.4.0-70-generic #78~18.04.1-Ubuntu SMP Sat Mar 20 14:10:07 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

leveson, 2021-03-25:
Hi. Thank you (:

The latest release fixes the first issue but I am still having trouble with the second which seems to be a problem on Ubuntu (and apparently its derivatives). Can you let me know if the following fixes the problem?

[bat.zip](https://github.com/leveson/bat/files/6203754/bat.zip)

lemurweb-dev, 2021-03-25:
That didn't solve the problem unfortunately

leveson, 2021-03-25:
Thank you for your help.
I will try to get ahold of an Ubuntu or Elementary OS distribution and comment again once I have pinned the issue down.

lemurweb-dev, 2021-03-25:
Hey i checked out your code and tried to understand what's wrong.
I guess in the last version you sent me you are creating two different services, bat-boot and bat-sleep.
Running `systemctl status bat-boot.service` returns this, i think it may be useful for you:
```
● bat-boot.service - Persist the battery charging threshold between restarts
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)

mar 25 15:23:49 catta-pc systemd[1]: bat-boot.service: Service has Restart= setting other than no, which isn't allowed for Type=oneshot services. Refusing.
mar 25 15:23:49 catta-pc systemd[1]: bat-boot.service: Service has Restart= setting other than no, which isn't allowed for Type=oneshot services. Refusing.
mar 25 15:34:48 catta-pc systemd[1]: bat-boot.service: Service has Restart= setting other than no, which isn't allowed for Type=oneshot services. Refusing.
mar 25 15:35:07 catta-pc systemd[1]: bat-boot.service: Service has Restart= setting other than no, which isn't allowed for Type=oneshot services. Refusing.
~

```

I guess it's only a matter of service config.
Let me know!

leveson, 2021-03-25:
What's the output when you run the same command on the other service?

lemurweb-dev, 2021-03-25:
Same,
```
● bat-sleep.service - Persist the battery charging threshold after hibernation
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)

mar 25 16:31:45 catta-pc systemd[1]: bat-sleep.service: Service has Restart= setting other than no, which isn't allowed for Type=oneshot services. Refusing.

```

leveson, 2021-03-25:
Thank you. That was really helpful.
Which version of systemd are you running? That can be found with `systemctl --version`.

lemurweb-dev, 2021-03-25:
```
catta@catta-pc:~$ systemctl --version
systemd 237
+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD -IDN2 +IDN -PCRE2 default-hierarchy=hybrid
```

leveson, 2021-03-25:
Seems to be an issue with systemd not allowing restarts for `oneshot` services but this has since been [fixed since v244-rc1](https://github.com/systemd/systemd/commit/d727acb6505743098aa1c8bb1c4c473b5fab1379).

lemurweb-dev, 2021-03-26:
Ok thank you for your time! I guess i'll have to wait for an elementaryOS update.
Closing the issue.

closed 2021-03-26 by lemurweb-dev