tshakalekholoane/bat #7: Not works in ASUS ROG GL552-VW

mojighahar, 2021-06-12:
Hi
I tested this tool on ASUS ROG GL552-VW and this message printed while setting the threshold:

```
$ sudo ./bat -t 80
This program is most likely not compatible with your system. See
https://github.com/leveson/bat#disclaimer.
```
Other commands like `-c` or `-s` works correctly.
I have ubuntu 18.10 and here is the output of `uname -a`:

```
Linux mojighahar 5.8.0-55-generic #62-Ubuntu SMP Tue Jun 1 08:21:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```

Please inform me if any other data required for clearing this issue. (output of `lshw` , ...)

leveson, 2021-06-12:
Run the following command and tell me if you get any output back.

```shell
sudo find /sys/ -name "charge_control*"
```

mojighahar, 2021-06-12:
It didn't find anything.
I do some relative search and find it:

```
$ sudo find /sys/ -name "charge*"
/sys/bus/platform/drivers/charger-manager
```

mojighahar, 2021-06-12:
Seems that it requires some extra kernel modules. Not?
If kernel upgrade may help, I can upgrade to ubuntu 21.04 and test it again.

mojighahar, 2021-06-12:
Here is content of `/sys/class/power_supply/BAT0` directory:

```
-rw-r--r-- 1 root root 4.0K Jun 12 21:44 alarm
-r--r--r-- 1 root root 4.0K Jun 12 09:53 capacity
-r--r--r-- 1 root root 4.0K Jun 12 21:44 capacity_level
-r--r--r-- 1 root root 4.0K Jun 12 21:44 cycle_count
lrwxrwxrwx 1 root root    0 Jun 12 21:44 device -> ../../../PNP0C0A:00
-r--r--r-- 1 root root 4.0K Jun 12 09:53 energy_full
-r--r--r-- 1 root root 4.0K Jun 12 09:53 energy_full_design
-r--r--r-- 1 root root 4.0K Jun 12 09:53 energy_now
drwxr-xr-x 3 root root    0 Jun 12 09:52 hwmon2
-r--r--r-- 1 root root 4.0K Jun 12 09:53 manufacturer
-r--r--r-- 1 root root 4.0K Jun 12 09:53 model_name
drwxr-xr-x 2 root root    0 Jun 12 21:33 power
-r--r--r-- 1 root root 4.0K Jun 12 09:53 power_now
-r--r--r-- 1 root root 4.0K Jun 12 09:53 present
-r--r--r-- 1 root root 4.0K Jun 12 09:53 serial_number
-r--r--r-- 1 root root 4.0K Jun 12 09:53 status
lrwxrwxrwx 1 root root    0 Jun 12 09:52 subsystem -> ../../../../../../../class/power_supply
-r--r--r-- 1 root root 4.0K Jun 12 09:53 technology
-r--r--r-- 1 root root 4.0K Jun 12 09:53 type
-rw-r--r-- 1 root root 4.0K Jun 12 09:52 uevent
-r--r--r-- 1 root root 4.0K Jun 12 09:53 voltage_min_design
-r--r--r-- 1 root root 4.0K Jun 12 09:53 voltage_now
```

leveson, 2021-06-12:
Upgrading might help.

closed 2021-06-19 by leveson