tshakalekholoane/bat #8: Can't attery status

mrunknownoffl, 2021-07-08:
Hi Leveson,

This works for my TUF series laptop but after I set to a value say 60, the status shows "unknown" instead of showing "Not charging". When I unplug, it shows as "Discharging".

My battery type is BAT1

Also, on Manjaro, I get some <failed> to start errors during startup. Please, let me know if there's anything to tinker with.

leveson, 2021-07-08:
I’m not sure I understand what the problem is. If you unplug your charger then the status has to rightfully show that your battery is discharging. The application merely relays what the operating system is telling it so if it says the charging status is unknown, then that is what it is reading. You mentioned it works anyway right?

What kind of start up issues are you experiencing on Manjaro?

mrunknownoffl, 2021-07-09:
Yeah it works.

Regarding manjaro , I get some <failed> to start the service during startup init screen. I don't find this on other distros though.

leveson, 2021-07-09:
Can you post the error message here? Does that mean it’s not working on Manjaro?

mrunknownoffl, 2021-07-10:
I'm sorry but I don't have Manjaro installed now. I'll open a new issue if I install anytime soon.

Thx for the script🙂

mrunknownoffl, 2021-07-15:
Hi, as I said earlier, the error is visible in manjaro because manjaro shows systemd services on startup. So, I am assuming that this is with any distro and not really an issue. From arch wiki, I got this for a similar implementation :

> This error is due to the service being started before asus-wmi could be loaded by the kernel (noted as kernel: battery: new extension: ASUS Battery Extension in the journal), making it impossible to write there.
The combination of those two options allows unlimited restart of the service for DefaultStartLimitIntervalSec (set to 10 seconds by default in /etc/systemd/system.conf), leaving enough time for the sysfs path to become available and writable and thus ensuring the fastest execution of the service.

But the thing is that battery starts to charge until I select manjaro linux in grub menu( I have windows and manjaro as dual boot) and stops when the os is actually started. So, anyway to stop charging right after starting the laptop i.e.: grub menu with this tool? Thanks.

leveson, 2021-07-15:
No. The operating system has to load and expose the variables before systemd can write to them again i.e. set the battery charging threshold, upon restart.

mrunknownoffl, 2021-07-15:
ok thanks :)

closed 2021-07-15 by mrunknownoffl