tshakalekholoane/bat #11: Issues with hibernate

igor-frolikov, 2021-07-25:
### Discussed in https://github.com/leveson/bat/discussions/10

<div type='discussions-op-text'>

<sup>Originally posted by **igor-frolikov** July 25, 2021</sup>
bat 0.7
works fine after restart and return from suspend, but fails after return from hibernate. Resumes working fine after restart.
I tried install from github prebuilt images and from aur via pamac - no any difference.

OS: Manjaro Linux [x86_64]
Host: VivoBook_ASUSLaptop X513IA_M513IA 1.0
Kernel: 5.10.52-1-MANJARO
Packages: 1277 (pacman)[stable]
Shell: bash 5.1.8
DE: KDE Plasma 5.22.3
WM: KWin (X11)
CPU: AMD Ryzen 5 4500U with Radeon Graphics (6) @ 2.375GHz
GPU: AMD ATI Renoir
Memory: 1720MiB / 15484MiB (11%)
Disk (/): 12GB / 100GB (12%)
Battery: 90% [Not charging]

PS.
Thank you very much for your efforts!</div>

leveson, 2021-07-25:
What do you get after running `bat --threshold` after resuming from hibernation?

igor-frolikov, 2021-07-26:
The threshold value remains the same, but battery keeps charging. If I set a threshold again all goes as it should.

[igor@mj ~]$ batstat
91
Charging
[igor@mj ~]$ bat-asus-battery -t
60
[igor@mj ~]$ bat60
[sudo] пароль для igor: 
Charging threshold set.
Use `sudo bat --persist` to persist the setting between restarts.
Persistence of the current threshold enabled.
[igor@mj ~]$ batstat
91
Not charging
[igor@mj ~]$ 
(I use aliases for bat-asus-battery commands)

igor-frolikov, 2021-07-26:
Might the info help? I did not try it yet due to my lack of knowledge.
https://wiki.archlinux.org/title/Laptop/ASUS#Battery_charge_threshold

leveson, 2021-07-26:
Let me know if this fixes the problem. Make sure it's the `bat-0.8-alpha` version. Set the threshold, restart your system, and then try to hibernate and see if persists afterwards.

Dealing with hibernation is a bit tricky because not every distribution implements hibernate target for systemd.

[bat.zip](https://github.com/leveson/bat/files/6880161/bat.zip)

igor-frolikov, 2021-07-26:
Great job! it works!

igor-frolikov, 2021-07-26:
Could you later update AUR with bat-0.8, please?

leveson, 2021-07-27:
Great! Thank you for filing the issue.

I don't maintain AUR packages, that's a community effort.

closed 2021-07-27 by leveson