tshakalekholoane/bat #12: Threshold not persisting after reboot - systemctl 245

mcarans, 2021-07-27:
After running `sudo bat -p` the threshold value after restart is back at 100.

I checked a related issue https://github.com/leveson/bat/issues/5. My version is greater than 244 you mentioned in there.

```
> systemctl --version
systemd 245 (245.4-4ubuntu3.11)
+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN +PCRE2 default-hierarchy=hybrid
```

When I look at the generated file I see that tehre is a question mark after the BAT instead of it saying 0:

```
> more /etc/systemd/system/bat-boot.service 
[Unit]
Description=Persist the battery charging threshold after boot 
After=multi-user.target
StartLimitBurst=0

[Service]
Type=oneshot
Restart=on-failure
ExecStart=/usr/bin/bash -c 'echo 60 > /sys/class/power_supply/BAT?/charge_control_end_threshold'

[Install]
WantedBy=multi-user.target
```

leveson, 2021-07-28:
Thank you for your feedback.

The question mark is a glob pattern that's supposed to capture any variation in the battery device name across different models.

The problem must be with the location of bash on your system. Can you confirm that the following works too before I start working on a fix for that?

 ```
[Unit]
Description=Persist the battery charging threshold after boot 
After=multi-user.target
StartLimitBurst=0

[Service]
Type=oneshot
Restart=on-failure
ExecStart=/bin/bash -c 'echo 60 > /sys/class/power_supply/BAT?/charge_control_end_threshold'

[Install]
WantedBy=multi-user.target
```

mcarans, 2021-07-28:
Thanks for looking into this. I tried changing the services to BAT? and did 
```
sudo systemctl daemon-reload

sudo systemctl restart bat-boot.service
```
There was no error so it seems to be only the bash problem.

leveson, 2021-07-30:
Thank you. The fix is up.

closed 2021-07-30 by leveson