tshakalekholoane/bat #19: I have an Asus laptop but I get an error - zenbook ux463fl

mr-tafreshi, 2021-08-18:
Hi.
I have an Asus laptop but I get an error:
This program is most likely not compatible with your system. See
https://github.com/leveson/bat#disclaimer.
model: zenbook flip ux46fl.
thanks.

leveson, 2021-08-18:
Can you provide more information about the operating system and the kernel version you are running?

The latter can be found by running the command `uname -r` in your terminal.

Also if you can confirm that you're not running it inside of a virtual machine.

mr-tafreshi, 2021-08-18:
Arch linux.
`uname -r` : `5.13.8-arch1-1`
Yes, I did not run it in Virtual Machine.

leveson, 2021-08-18:
Run the following and let me know if you get anything. 

```sh
sudo find /sys/ -name "charge_control*"
```

If that doesn’t work, check to see if you have a directory with the prefix BAT in `/sys/class/power_supply/`. It’s usually something like `BAT0` or `BATT` on newer systems.

mr-tafreshi, 2021-08-19:
[image]

leveson, 2021-08-19:
The first command requires elevated permissions, that's why you're getting errors. Prepend `sudo` to the command just as it appears in my comment above.

You seem to have the variable so let's see what options it exposes by listing its contents

```sh
ls /sys/class/power_supply/BAT0/
```

mr-tafreshi, 2021-08-19:
[image]

leveson, 2021-08-20:
Your system does not seem to have the variable required and it's hard to pin point why. It could be anything from your distribution to an outdated BIOS. Both of which seem unlikely. Or it could be that on newer models, the variable is located under a different address and this hasn't been captured in the Linux driver repository.

You could try first updating your BIOS firmware, then if that doesn't work, another primary distribution like Debian (Experimental or Unstable).