tshakalekholoane/bat #20: issue: Charging does not stop no matter if the laptop is running or sleeping

seapat, 2021-10-05:
### Subject of the issue

The charging seems to not stop at the intended threshold despite correct output from the program (I did not receive an error, like on an older laptop that probably just does not support the feature)

I am using KDE, and I am not really sure if it hibernates or suspends when I close the lid (the settings only have one option: "sleep"). I am also judging only by `bat -c` and the display in in the KDE System Tray, maybe this is just a display issue.

### Your system

OS: Fedora 35 KDE Linux x86_64
Host: VivoBook Flip 14 (J412U)
Kernel: 5.14.9-200.fc34
systemctl 248

### Version

0.8.3 (Latest)

### Steps to reproduce

What I did:
```
bat -t 65
bat -p
bat -t
>> 65
```

what I've tried after looking at #3 
`test -f /etc/systemd/system/bat.service && echo "TRUE"` returns nothing

`test -f /sys/class/power_supply/BAT?/charge_control_end_threshold && echo "TRUE"` returns `TRUE`

`systemctl status bat.service` returns `Unit bat.service could not be found.`

`systemctl status bat-sleep.service` returns:
```
bat-sleep.service - Persist the battery charging threshold after sleep
     Loaded: loaded (/etc/systemd/system/bat-sleep.service; enabled; vendor preset: disabled)
     Active: inactive (dead)

Oct 05 15:25:31 fedora systemd[1]: Finished Persist the battery charging threshold after sleep.
Oct 05 15:57:43 fedora systemd[1]: Starting Persist the battery charging threshold after sleep...
Oct 05 15:57:43 fedora systemd[1]: bat-sleep.service: Deactivated successfully.
Oct 05 15:57:43 fedora systemd[1]: Finished Persist the battery charging threshold after sleep.
Oct 05 15:59:14 fedora systemd[1]: Starting Persist the battery charging threshold after sleep...
Oct 05 15:59:14 fedora systemd[1]: bat-sleep.service: Deactivated successfully.
Oct 05 15:59:14 fedora systemd[1]: Finished Persist the battery charging threshold after sleep.
Oct 05 16:02:51 fedora systemd[1]: Starting Persist the battery charging threshold after sleep...
Oct 05 16:02:51 fedora systemd[1]: bat-sleep.service: Deactivated successfully.
Oct 05 16:02:51 fedora systemd[1]: Finished Persist the battery charging threshold after sleep.
```

### Expected behaviour

charging should stop at the threshold both when the lid is closed and when the device is running.

### Actual behaviour

Both with closed lid or running system, the charging continues beyond the set limit. Shutdown and restart also did not have an effect on this. `bat -t` still reports the threshold I set correctly after restart.

I am clueless when it comes to systemd but judging by the output above it seems that

tshakalekholoane, 2021-10-06:
I am puzzled because by the looks of things everything seems to be working. 

At least it should because for example, the output of the `bat -t` command for is read directly from the OS and if that's the output you are getting then theoretically everything should be fine.

Perhaps you could try a BIOS update and let me know if that works?

This is inherently difficult to debug because of all the differences between manufacturers, models, BIOS versions, and the fragmented Linux distribution ecosystem.

seapat, 2021-10-10:
Hey,

I got around to setting up the systemd service. This did not work as well, so I looked around for solutions related to fedora and found this [blog post](https://blog.nil.im/?7b) which seems to work and be persistent after reboot. The relevant part is the following:

>  Just create a file /etc/tmpfiles.d/asus_battery.conf and write the following into it:
`w /sys/class/power_supply/BAT0/charge_control_end_threshold - - - - 60`

Which seems to be to have done the trick. It  still is a little weird though, since `/sys/class/power_supply/BAT0/status` is still "Unknown" when at the threshold I set. The blog post also mentions that the threshhold could be overwritten using something like `echo 100 | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold`. This does not work for me somehow.

Btw, this is what I get from `ls sys/class/power_supply/`:

`ADP1  BAT0  hid-0018:04F3:2766.0002-battery`

So there's a third device called "hid-0018:04F3:2766.0002-battery" with "/sys/class/power_supply/hid-0018\:04F3\:2766.0002-battery/status" being "Unknown". `acpi` shows the following:

```
Battery 0: Unknown, 0%, rate information unavailable
Battery 1: Unknown, 60%
```
Apparently that can be caused by some other connected device. Since I don't have that it might be caused by something internal of the laptop but I'm not sure

tshakalekholoane, 2021-10-11:
I find it a bit weird because all that does is write the charge control threshold when the system boots in the same way piping using `echo 100 | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold` or `bat -t 60` does (or at least should) according to the [documentation](https://www.freedesktop.org/software/systemd/man/tmpfiles.d.html).

If it works when you set it using the configuration file then maybe its a permissions issue? If you have root access then perhaps running `bat -t 60` using that might help. But then it could also be an issue with how Fedora is configured? I honestly have no idea or why it's showing multiple devices plugged in.