tshakalekholoane/bat #21: issue: How to set charge resume level?

Aurosutru, 2021-11-07:
Hi,

This app is working nicely on an Asus M15 laptop running Fedora 34.

I have set the upper charge threshold at 60% and am wondering at what point charging will resume.  One battery manager uses a low 30% figure for resuming charging and another uses 58%.  Is there a way to adjust the resume charging level using bat?

BTW, the Fedora charging indicator correctly shows "Not Charging" when battery is above 60% charge.

Thanks.

tshakalekholoane, 2021-11-08:
> Is there a way to adjust the resume charging level using bat?

No, there is no way to adjust the resume charging level using bat. All it does is set the threshold level.

Aurosutru, 2021-11-08:
Does the charging resume as soon as the battery level goes below 60%, or whatever number is set as the  threshold?

tshakalekholoane, 2021-11-08:
Yes, that is the default behaviour on most systems.

closed 2021-11-08 by Aurosutru