tshakalekholoane/bat #22: issue: Fails to build

ElectricRCAircraftGuy, 2021-12-17:
`go build ./cmd/bat/` results in 

>     build tshaka.co/bat/cmd/bat: cannot load embed: malformed module path "embed": missing dot in first path element

Steps to reproduce on Ubuntu 20.04:
```bash
git clone https://github.com/tshakalekholoane/bat.git
cd bat
sudo apt install golang-go
time go build ./cmd/bat/
```

tshakalekholoane, 2021-12-17:
Which version of Go are you running?

ElectricRCAircraftGuy, 2021-12-17:
> Which version of Go are you running?

```bash
$ go version
go version go1.13.8 linux/amd64
```

tshakalekholoane, 2021-12-17:
Embedded files were only introduced in [Go 1.16](https://go.dev/doc/go1.16#library-embed). Try with a version later than that if you  prefer building over grabbing a binary from the releases page.

closed 2021-12-22 by tshakalekholoane