tshakalekholoane/bat #25: issue: bat uninstall

batgithubpf, 2022-04-26:
### Subject of the issue

Bat is easy to install and works fine in ubuntu/asus: how to uninstall it?

### Your system

OS:  Ubuntu 22.04 Linux x86_64
Host: ZenBook 
Kernel: 5.15.0-27-generic

### Version

0.8.4 (Latest)

tshakalekholoane, 2022-04-26:
Just delete the `bat` file.

batgithubpf, 2022-04-27:
Not so simple!

-------- Message initial --------

Répondre à: tshakalekholoane/bat
À: tshakalekholoane/bat ***@***.***>
Cc: batgithubpf ***@***.***>, Author
Date: Tue, 26 Apr 2022 11:14:30 -0700

gamegine, 2022-04-27:
run --reset first to remove the changes introduced by --persist and then remove the executable bat file

batgithubpf, 2022-04-27:
It's OK, thank you.

-------- Message initial --------

Date: Wed, 27 Apr 2022 00:31:01 -0700

tshakalekholoane, 2022-04-27:
Did that solve the problem? Running the following as @gamegine suggested is usually enough since the entire application is contained in a single binary but I could be missing something.

```
bat --reset && rm bat
```

closed 2022-05-02 by tshakalekholoane

batgithubpf, 2022-10-11:
Yes thank you, seems to have solved the problem.
-------- Message initial --------

Date: Wed, 27 Apr 2022 05:47:35 -0700