tshakalekholoane/bat #27: bat-asus-battery-bin package on AUR failed to build

duymazino, 2022-06-13:
### Subject of the issue

Latest AUR package of 0.9.1-1 is unable to build. Package manager returned the package as unable to build because one or more files did not pass validity check. 

Pamac log:

```
Building bat-asus-battery-bin...
==> Making package: bat-asus-battery-bin 0.9.1-1
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found bat.zip
==> Validating source files with sha256sums...
    bat.zip ... FAILED
==> ERROR: One or more files did not pass the validity check!
Failed to build bat-asus-battery-bin
```

Please confirm and push a fix. Thank you.

### Your system

OS: Manjaro Linux x86_64 
 
Kernel: 5.15.41-1-MANJARO 

Shell: zsh 5.9 

DE: GNOME 42.2 

### Version

0.9.1 (Latest)

### Steps to reproduce

Build in Pamac from AUR package.

### Expected behaviour

Successful build.

### Actual behaviour

Unable to build

tshakalekholoane, 2022-06-13:
The AUR package is a community effort. I don't manage that so I cannot help you with that.

duymazino, 2022-06-13:
> 

Got it. I'll forward it to the current packager then. Good bye.