tshakalekholoane/bat #42: Request procedure how to update the app

13thRavenPurple, 2023-02-25:
Hi! First of all i would like to Thank you for this great app. Working extremely well on my VivoBook_ASUSLaptop X415JA 1.0 running Linux Mint 21.1 x86_64 using kernel 5.15.0-58-generic with DE of Xfce 4.16

Unfortunately, i could not update this app. Downloading the latest version and applying the required permission and then substituting it to the older version does not work for me. Giving a guide or way how to update is much appreciated. I am currently running bat 0.10.1-0-gc5871eb-dirty version by the way.

tshakalekholoane, 2023-02-25:
You're welcome @13thRavenPurple (:

I am not sure I understand your problem. Upgrading is just as simple as deleting the previous version and downloading the most recent. Unless you're building it from source but you're not doing that are you?

13thRavenPurple, 2023-02-27:
> I am not sure I understand your problem. Upgrading is just as simple as deleting the previous version and downloading the most recent. 

I tried again but it still failed =(
```
milla@DiggieIsBack:~$ bat -v
bat 0.10.1-0-gc5871eb-dirty
Copyright (c) 2023 Tshaka Eric Lekholoane.
MIT Licence.
```
also i tried
`bat --reset && rm bat`
as stated here https://github.com/tshakalekholoane/bat/issues/25
but it also failed =(

> Unless you're building it from source but you're not doing that are you?

No. i used the  chmod one.

tshakalekholoane, 2023-02-28:
I can't think of why it isn't working. Since you're using a desktop environment, what happens when you delete it using your file explorer? 

The reset command also requires elevated permissions so it's also possible that if you're running the above command as is, `rm bat` isn't being executed because the `bat reset` is failing. In which you could try to run them separately as follows and see if that is indeed the case.

```bash
sudo bat reset
rm bat # Possibly with sudo too if that doesn't work 🤷.
```

13thRavenPurple, 2023-03-01:
Hi Thank you again for your assistance 👍 I finally manage to update it 🍻 the reason for the hold up is that there is a copy in `/usr/local/bin` aside from `~/` ... maybe i misunderstood your instruction at https://github.com/tshakalekholoane/bat#installation 😅 for after reinstalling with the latest version, the app would not run unless it is on `/usr/local/bin`

```
milla@DiggieIsBack:~$ bat --version
bat 0.12-0-g8daf723
Copyright (c) 2023 Tshaka Eric Lekholoane.
MIT Licence.
```

Again Thank you so much for helping me out and Sorry for the trouble. Wish you all the Best and More Power 🍻

closed 2023-03-01 by 13thRavenPurple