tshakalekholoane/bat #65: issue: Battery percentage setting doesn't persist; systemctl does not indicate that bat@.service is running.

cloudmadeofcandy, 2023-09-17:
### Subject of the issue

Hi, I've downloaded and used the bat file from the Release page, version 0.13, following your instruction to the letter but the setting doesn't persist after restarting.

When I run the following command in the Terminal:
```
systemctl --state=active --type=service
```
I do not see ```bat@.service``` running.

[image]

However, when I check for all services:
```
systemctl --type=service
```
I see that there's a service ```bat@multi-user.service```
[image]

The service file is identical to that in your source code.

### Your system

OS: Ubuntu 22.04.3 LTS x86_64 
Host: ZenBook UX425EA_UX425EA 1.0 
Kernel: 6.2.0-32-generic 
Shell: bash 5.1.16 
Resolution: 1920x1080 
DE: Unity 7.6.0 
WM: Compiz 
Terminal: gnome-terminal

### Version

0.13 (Latest)

### Steps to reproduce

Download and run the bat file as per instructions, then restart/shutdown the laptop.

### Expected behaviour

After restart/turn on, the service should automatically be activated by the systemd and the systemctl.

Running ```bat threshold``` should return the pre-determined value.

### Actual behaviour

The service isn't run, not after restarting. Running ```bat threshold```  returns the default value (100).

pepa65, 2023-09-17:
On your system, what is the output of `cat /etc/systemd/system/bat-*`

cloudmadeofcandy, 2023-09-17:
> On your system, what is the output of `cat /etc/systemd/system/bat*`

It is identical to the source file. 

```
[Unit]
Description=Persist the battery charging threshold after %i
After=%i.target
StartLimitBurst=0

[Service]
Type=oneshot
Restart=on-failure
ExecStart=%s -c 'echo 60 > /sys/class/power_supply/BAT?/charge_control_end_threshold'

[Install]
WantedBy=%i.target
```

There really is no difference.

I also notice that the battery limiter also works after shutting down, however when I turn it on, the threshold immediately resets to 100

pepa65, 2023-09-17:
What is `%s`?
And what is `cat /sys/class/power_supply/BAT?/charge_control_end_threshold`??

cloudmadeofcandy, 2023-09-17:
> What is %s?
I don't really know how to print that out, however I believe that it should be the location of the /bin/bash.

>And what is cat /sys/class/power_supply/BAT?/charge_control_end_threshold??
60. After I run the bat file. After restarting, it is definitely 100.

pepa65, 2023-09-17:
The unit file should not have %i and %s, but the substituted values. If you do `cat /etc/systemd/system/bat*` it should print `/bin/bash` and not %s.

cloudmadeofcandy, 2023-09-17:
How very interesting. However, the %s and %i must be substituted correctly, otherwise it wouldn't have run in the first place.

A picture of the ```cat /etc/systemd/system/bat*``` tho:

[image]

pepa65, 2023-09-17:
That is not going to run like that. When I do `systemctl list-unit-files` I see my bat with `enabled enabled` both in green. I am guessing that is not the case with you. I bet if you replace `%s` with `/bin/bash` and `%i` with `multi-user` it will work on reboot.

cloudmadeofcandy, 2023-09-17:
Alright, I'll try that.

cloudmadeofcandy, 2023-09-17:
@pepa65 you are correct.

After I turned the laptop on, the battery indicator turned red again, but when the ubuntu shell appeared, the battery indicator turned white again. This is a most peculiar turn of event. 

The ```bat@.service``` is still ```indirect enabled```, tho. What is happening and how do I fix it? The glob should work, right?

Should I close the issue when it is not fully solved?

pepa65, 2023-09-17:
If you did `bat persist` I would expect for it to work correctly. Let's wait for @tshakalekholoane to chime in, and see if there is something wrong with it.

(My fork is a bit different, and then this repo got changed as well...)

cloudmadeofcandy, 2023-09-17:
after I commanded ```sudo bat persist```, the contents of the ```bat@.service``` reverted back to the previous version. When I try to enable the service with systemctl, this is what it said:

```
Failed to get properties: Unit name bat@.service is neither a valid invocation ID nor unit name.
```

pepa65, 2023-09-17:
That confirms there is something wrong that @tshakalekholoane needs to fix.

pepa65, 2023-09-18:
@tshakalekholoane It is also good to include `RemainAfterExit=true` in the unit files under `[Service]`, because then the service shows as Active when doing `systemctl status`.

tshakalekholoane, 2023-09-18:
I suspect it may have something to do with the unit file not being in the correct format i.e., the target cannot be specified using a template (I think). 

The latest version uses systemd templates (see `man systemd.unit`) instead of writing all the units manually. The thinking being this would allow services to be created for events that are only supported by the distribution and the `%s` specifier would also help with finding the shell on distributions like NixOS that do use standard paths.

The fix may well be going back to the previous implementation.

tshakalekholoane, 2023-09-18:
I just pushed a fix, @cloudmadeofcandy. Please let me know if it works. 

It also includes @pepa65's suggestion to set `RemainAfterExit=true` so the service status can now be queried.

[bat.zip](https://github.com/tshakalekholoane/bat/files/12652495/bat.zip)

pepa65, 2023-09-19:
When I checked out the repo and built, I got `bat 0.13-3-g7f0908d` and it was as before. When I downloaded the `bat.zip` I got `bat 0.13-4-g32b7edc` and it worked for the unit file path expansion, but the `%s` was still there instead of `/bin/bash`.

tshakalekholoane, 2023-09-19:
I haven’t merged the changes yet. They’re in [this PR](https://github.com/tshakalekholoane/bat/pull/67) if you want to checkout the code. 

Yes I left the `%s` in there so that the application doesn’t have to contain logic to find the shell. systemd can take care of that. Does it not work for you?

pepa65, 2023-09-19:
Oh, I didn't know, does systemd do that, replace %s with a valid shell? In my fork, I just use /bin/sh as I assume that to be universally available on Linux.

cloudmadeofcandy, 2023-09-19:
After uninstalling the old bat files in ```/etc/systemd/system``` and reloaded the daemon with ```systemctl```, I reinstalled the new bat binary, set and persisted the threshold and restarted the laptop. It didn't work.

[image]

When I ran

```
systemctl --type=service
```

the system reported that ```bat@multi-user.service```'s status is ```failed failed```
[image]

cloudmadeofcandy, 2023-09-19:
The result after running ```cat /etc/systemd/system/bat* ```:

```
[Unit]
Description=Persist the battery charging threshold after hibernate
After=hibernate.target
StartLimitBurst=0

[Service]
Type=oneshot
ExecStart=%s -c 'echo 60 > /sys/class/power_supply/BAT?/charge_control_end_threshold'
Restart=on-failure
RemainAfterExit=true

[Install]
WantedBy=hibernate.target
[Unit]
Description=Persist the battery charging threshold after hybrid-sleep
After=hybrid-sleep.target
StartLimitBurst=0

[Service]
Type=oneshot
ExecStart=%s -c 'echo 60 > /sys/class/power_supply/BAT?/charge_control_end_threshold'
Restart=on-failure
RemainAfterExit=true

[Install]
WantedBy=hybrid-sleep.target
[Unit]
Description=Persist the battery charging threshold after multi-user
After=multi-user.target
StartLimitBurst=0

[Service]
Type=oneshot
ExecStart=%s -c 'echo 60 > /sys/class/power_supply/BAT?/charge_control_end_threshold'
Restart=on-failure
RemainAfterExit=true

[Install]
WantedBy=multi-user.target
[Unit]
Description=Persist the battery charging threshold after suspend
After=suspend.target
StartLimitBurst=0

[Service]
Type=oneshot
ExecStart=%s -c 'echo 60 > /sys/class/power_supply/BAT?/charge_control_end_threshold'
Restart=on-failure
RemainAfterExit=true

[Install]
WantedBy=suspend.target
[Unit]
Description=Persist the battery charging threshold after suspend-then-hibernate
After=suspend-then-hibernate.target
StartLimitBurst=0

[Service]
Type=oneshot
ExecStart=%s -c 'echo 60 > /sys/class/power_supply/BAT?/charge_control_end_threshold'
Restart=on-failure
RemainAfterExit=true

[Install]
WantedBy=suspend-then-hibernate.target

```

My service folder now have multiple ```bat@.service``` files, and they aren't sorted into folders like the previous version:

[image]

tshakalekholoane, 2023-09-19:
> I just use /bin/sh as I assume that to be universally available on Linux.

I’m not sure that’s the case on NixOS but that could be a good default. 

@cloudmadeofcandy are you sure you’re running the correct version? `bat@multi-user` shouldn’t be there anymore. There should be 5 different services for each event after the change.

tshakalekholoane, 2023-09-19:
What’s their status after restart?

cloudmadeofcandy, 2023-09-19:
> @cloudmadeofcandy are you sure you’re running the correct version? 

The version of the software said "18 September 2023"

[image]

pepa65, 2023-09-19:
Best disable & delete those old `bat@*` unit files, the new files start with `bat-`.

cloudmadeofcandy, 2023-09-19:
Before the reboot, after setting and persisting:

[image]

After the reboot, querying the threshold:

[image]

There is only a service called ```bat-multi-user.service```, in ```activating auto-restart```

[image]

cloudmadeofcandy, 2023-09-19:
> Best disable & delete those old bat@* unit files, the new files start with bat-.

I did, but to no avail...

pepa65, 2023-09-19:
So you still have `%s` in your unit file(s)? Check if replacing that with `/bin/sh` helps make it work. I am suspecting that systemd doesn't (always?) replace `%s` with a default shell.

cloudmadeofcandy, 2023-09-19:
replacing ```%s``` with  ```/bin/sh``` doesn't work. I've just tried it 
[image]

cloudmadeofcandy, 2023-09-19:
replacing ```%s``` with ```/bin/bash``` does though.

pepa65, 2023-09-19:
I guess the syntax in that line doesn't work for `dash`... I remember I adapted it on my fork.

cloudmadeofcandy, 2023-09-19:
> Which version of systemd are you using? 

Mine is systemd 249 (249.11-0ubuntu3.10)

cloudmadeofcandy, 2023-09-19:
> But either way, maybe resolving it manually could be the solution here too.

True, but whenever I want to change the threshold, I'd have to edit the file again.

I'll close the issue in the mean time.

pepa65, 2023-09-19:
@tshakalekholoane You could adjust the line as I suggested, and it would work in either `bash` or `dash` or whatever POSIX shell. I am testing a NixOS installation, but the NixOS live CD had `/bin/sh` linked to one of their binaries.

tshakalekholoane, 2023-09-19:
But according to https://github.com/tshakalekholoane/bat/issues/65#issuecomment-1725016431 `/bin/sh` also doesn't work so I'd rather mark Bash as a hard requirement instead of trying to enumerate all the possibilities.

@cloudmadeofcandy, please let me know if the following works.
 
[bat.zip](https://github.com/tshakalekholoane/bat/files/12658531/bat.zip)

pepa65, 2023-09-19:
I am pretty sure NixOS also has `/bin/sh`. But it would require a change in your line to make it work in `dash`.

tshakalekholoane, 2023-09-19:
Yes, but being immutable and all, does it allow changing the threshold manually or otherwise?

cloudmadeofcandy, 2023-09-19:
> @cloudmadeofcandy, please let me know if the following works.

This works perfectly. Thank you so much :100:

closed 2023-09-19 by cloudmadeofcandy

pepa65, 2023-09-19:
> Yes, but being immutable and all, does it allow changing the threshold manually or otherwise?

I would hope so! I think it's more the software that's immutable than the system variables. Otherwise there would be no point to use `bat`, and no use for `bat` to support NixOS.

pepa65, 2023-09-21:
I can now report that NixOS does have `/bin/sh` (it is the only entry they have in `/bin`, a link to their store for the bash shell). That said, POSIX does not require that, but they do require `sh` to be present in PATH, so if you want to be supersafe, you could use your approach to find `bash`, but instead use it to find `sh` (as `bash` is not guaranteed to be present).

tshakalekholoane, 2023-09-22:
Maybe. But I’m still reluctant to use `bin/sh` given for whatever reason that doesn’t seem to work on Ubuntu (per comment above).

Were you able to determine if NixOS also allows the charging threshold to be set?

pepa65, 2023-09-22:
That line doesn't work because `sh` doesn't expand the question mark in `echo 60 >/sys/class/power_supply/BAT?/charge_control_end_threshold`. You have to do: `b=$(echo /sys/class/power_supply/BAT?/charge_control_end_threshold); echo 80 >"$b"` (or you have to substitute the actual value into the unit files).

tshakalekholoane, 2023-09-22:
Ah. I see. Then it might better to do in the unit file template since the setup logic already resolves the actual value.

commit b3203946 (2023-09-19): fix: restore persistence functionality

This commit addresses an issue where the switch to using systemd
templates and format specifiers introduced a regression, causing the
persistence functionality to no longer work as expected.

To resolve this issue, we have reverted to an earlier implementation
that manually writes all the units and uses Go's exec.LookPath to
resolve the shell path.

Resolves: #65
PR #67 (2023-09-19): fix: restore persistence functionality
This addresses an issue where the switch to using systemd templates and format specifiers introduced a regression, causing the persistence functionality to no longer work as expected.

To resolve this issue, we have reverted to an earlier implementation
that manually writes all the units and uses Go's `exec.LookPath` to
resolve the shell path.

Resolves: #65
commit b1269666 (2023-09-19): fix: restore persistence functionality

This commit addresses an issue where the switch to using systemd
templates and format specifiers introduced a regression, causing the
persistence functionality to no longer work as expected.

To resolve this issue, we have reverted to an earlier implementation
that manually writes all the units and uses Go's exec.LookPath to
resolve the shell path.

Resolves: #65