tshakalekholoane/bat #70: Not working on Asus Zenbook with Ubuntu 23.10

deepaksp, 2023-09-25:
### Subject of the issue

Its not working, it keep charging full.

### Your system

For example,

OS: Manjaro Linux x86_64
Host: ZenBook UX425EA_UX425EA 1.0
Kernel: 5.13.8-1-MANJARO

### Version

0.13.1 (Latest)

deepaksp, 2023-09-25:
i ran
sudo bat threshold 80
and then sudo bat persist.

but it keep charging beyond 80%

pepa65, 2023-09-25:
What is the output of `grep -H [0-9] /sys/class/power_supply/BAT?/charge_control_end_threshold` and what of `lsmod |grep asus`?

deepaksp, 2023-09-26:
lsmod |grep asus

asus_nb_wmi            28672  0
asus_wmi               73728  2 asus_nb_wmi,mfd_aaeon
ledtrig_audio          16384  3 snd_ctl_led,snd_hda_codec_generic,asus_wmi
sparse_keymap          16384  1 asus_wmi
platform_profile       16384  1 asus_wmi
video                  69632  2 asus_wmi,i915
wmi                    40960  4 video,asus_wmi,wmi_bmof,mfd_aaeon

grep -H [0-9] /sys/class/power_supply/BAT?/charge_control_end_threshold
/sys/class/power_supply/BAT0/charge_control_end_threshold:80

pepa65, 2023-09-26:
That all looks good... I am not sure if the hardware doesn't obey the software sometimes, but I have had that only when the current charge is still lower than the threshold that it works. Also, when I set any limit below 53, it will just charge, seemingly ignoring the limit.

It is worth checking out if there are some additional/limiting settings in the BIOS.

Try draining the battery until in the 70s, and then plugging the power back in, while the charge threshold is set to 80. See if it works then.

tshakalekholoane, 2023-09-29:
This is puzzling indeed. You can try the latest version and see if that fixes anything. But from the output you sent, it seems like it should be working.

closed 2023-09-29 by tshakalekholoane