tshakalekholoane/bat #77: issue: charge threshold not working after waking up from sleep

HarshDobariya79, 2023-11-27:
### Subject of the issue

Charge threshold persistence not working after sleep

### Your system

OS: Ubuntu 23.10 (mantic)
Host: Asus TUF A15 FA506IH
Kernel: 6.5.0-13-generic

### Version

0.14 (Latest)

### Steps to reproduce

- set charge threshold
- persist the charge threshold
- send the host machine to sleep
- wake up the machine

### Expected behaviour

The persisted threshold should work

### Actual behaviour

The charge threshold is not working at all unless set again

pepa65, 2023-11-27:
How do you "send the host machine to sleep"?

In any case, I think it might be because `sleep` is used in the persist settings. To test if this is the case, do `wget good4.eu/bat; chmod +x bat; sudo ./bat persist` and see if that fixes the issue.

HarshDobariya79, 2023-11-27:
Also the threshold works fine while the machine is in sleep and stops working as soon as the machine wakes up.

pepa65, 2023-11-27:
Reloading the driver resets the threshold value. The unit files `/etc/systemd/system/bat-*` together with `/etc/systemd/system/*.wants/bat-*` should take care of that.

What does `/var/log/system` look like right after coming back from sleep. What are the lines after the line `System returned from sleep state.`, do you have `Starting Persist the battery charging threshold after suspend...`?

HarshDobariya79, 2023-11-27:
**/etc/systemd/system/bat-hybrid-sleep.service**

```
[Unit]
Description=Persist the battery charging threshold after hybrid-sleep
After=hybrid-sleep.target
StartLimitBurst=0

[Service]
Type=oneshot
ExecStart=/usr/bin/sh -c 'echo 60 > /sys/class/power_supply/BAT1/charge_control_end_threshold'
Restart=on-failure
RemainAfterExit=true

[Install]
WantedBy=hybrid-sleep.target
```

**/etc/systemd/system/hybrid-sleep.target.wants/chargelimit-hybrid-sleep.service**

```
[Unit]
Description=Persist the battery charge limit after hybrid-sleep
After=hybrid-sleep.target
StartLimitBurst=0

[Service]
Type=oneshot
ExecStart=/usr/bin/sh -c 'echo 60 >/sys/class/power_supply/BAT1/charge_control_end_threshold'
Restart=on-failure
RemainAfterExit=true

[Install]
WantedBy=hybrid-sleep.target
```

**No trace of battery persistent in syslog after waking up.**

pepa65, 2023-11-27:
So in `/var/log/syslog` after the line `System returned from sleep state.` you must have soon after that: `Reached target Suspend.`. On my system I have right after that `Starting Persist the battery charging threshold after suspend...`.

What does `sudo systemctl status bat-suspend.service` give?

HarshDobariya79, 2023-11-27:
There is no trace of `Starting Persist the battery charging threshold after suspend...`, I also tried to search in the logs file. The `bat-suspend.service` is active.

**/var/log/syslog**

```
2023-11-27T15:01:03.013980+05:30 23 systemd[1]: Reached target suspend.target - Suspend.
2023-11-27T15:01:03.015894+05:30 23 ModemManager[1297]: <info>  [sleep-monitor-systemd] system is resuming
2023-11-27T15:01:03.046468+05:30 23 kernel: [15215.439762] Generic FE-GE Realtek PHY r8169-0-200:00: attached PHY driver (mii_bus:phy_addr=r8169-0-200:00, irq=MAC)
2023-11-27T15:01:03.062648+05:30 23 NetworkManager[1089]: <info>  [1701077463.0155] manager: sleep: wake requested (sleeping: yes  enabled: yes)
2023-11-27T15:01:03.063293+05:30 23 NetworkManager[1089]: <info>  [1701077463.0158] device (enp2s0): state change: unmanaged -> unavailable (reason 'managed', sys-iface-state: 'external')
2023-11-27T15:01:03.063529+05:30 23 systemd[1]: Starting grub-common.service - Record successful boot for GRUB...
2023-11-27T15:01:03.067127+05:30 23 systemd[1]: Starting nvidia-resume.service - NVIDIA system resume actions...
2023-11-27T15:01:03.071577+05:30 23 suspend: nvidia-resume.service
2023-11-27T15:01:03.071714+05:30 23 logger[88302]: <13>Nov 27 15:01:03 suspend: nvidia-resume.service
2023-11-27T15:01:03.074870+05:30 23 systemd[1]: Stopped target suspend.target - Suspend.
2023-11-27T15:01:03.080368+05:30 23 systemd[1]: nvidia-resume.service: Deactivated successfully.
2023-11-27T15:01:03.081143+05:30 23 systemd[1]: Finished nvidia-resume.service - NVIDIA system resume actions.
2023-11-27T15:01:03.087695+05:30 23 systemd[1]: grub-common.service: Deactivated successfully.
2023-11-27T15:01:03.088092+05:30 23 systemd[1]: Finished grub-common.service - Record successful boot for GRUB.
2023-11-27T15:01:03.091526+05:30 23 systemd[1]: Starting grub-initrd-fallback.service - GRUB failed boot detection...
2023-11-27T15:01:03.102964+05:30 23 systemd[1]: grub-initrd-fallback.service: Deactivated successfully.
2023-11-27T15:01:03.103600+05:30 23 systemd[1]: Finished grub-initrd-fallback.service - GRUB failed boot detection.
2023-11-27T15:01:03.187569+05:30 23 gnome-shell[3264]: Window manager warning: Overwriting existing binding of keysym 35 with keysym 35 (keycode e).
2023-11-27T15:01:03.187764+05:30 23 gnome-shell[3264]: Window manager warning: Overwriting existing binding of keysym 31 with keysym 31 (keycode a).
2023-11-27T15:01:03.187816+05:30 23 gnome-shell[3264]: Window manager warning: Overwriting existing binding of keysym 32 with keysym 32 (keycode b).
2023-11-27T15:01:03.187877+05:30 23 gnome-shell[3264]: Window manager warning: Overwriting existing binding of keysym 33 with keysym 33 (keycode c).
2023-11-27T15:01:03.187922+05:30 23 gnome-shell[3264]: Window manager warning: Overwriting existing binding of keysym 34 with keysym 34 (keycode d).
```

**bat-suspend.service**

```
● bat-suspend.service - Persist the battery charging threshold after suspend
     Loaded: loaded (/etc/systemd/system/bat-suspend.service; enabled; preset: enabled)
     Active: active (exited) since Mon 2023-11-27 09:23:54 IST; 5h 45min ago
   Main PID: 4447 (code=exited, status=0/SUCCESS)
        CPU: 5ms

Nov 27 09:23:54 23.10 systemd[1]: Starting bat-suspend.service - Persist the battery charging threshold after suspend...
Nov 27 09:23:54 23.10 systemd[1]: Finished bat-suspend.service - Persist the battery charging threshold after suspend.
```

pepa65, 2023-11-27:
All this would not be necessary if the driver would just accept a parameter to set the charge limit...

We need a systemd expert to get this to work I guess. Side note, I am on 22.04, you seem to be on 23.10, so the systemd version is different.

HarshDobariya79, 2023-11-27:
> All this would not be necessary if the driver would just accept a parameter to set the charge limit...
> 
> We need a systemd expert to get this to work I guess. Side note, I am on 22.04, you seem to be on 23.10, so the systemd version is different.

Yes. I was also using it on **22.04 Jammy** and it was working perfectly fine.

HarshDobariya79, 2023-11-27:
I found this service inactive, but I can't enable it.

```
○ chargelimit-suspend.service - Persist the battery charge limit after suspend
     Loaded: loaded (/etc/systemd/system/chargelimit-suspend.service; enabled; preset: enabled)
     Active: inactive (dead)
```

pepa65, 2023-11-27:
> I found this service inactive, but I can't enable it.

What you can try is `bat reset` (with the installed binary) and then `sudo ./bat remove` (with the downloaded binary) followed by `sudo ./bat persist`. Then see what the output is of `./bat`.

pepa65, 2023-11-27:
There must be something in the requirements for the systemd unit files that changed so they no longer work in the 23.10 version...

HarshDobariya79, 2023-11-27:
> There must be something in the requirements for the systemd unit files that changed so they no longer work in the 23.10 version...

Yes, most likely.

pepa65, 2023-11-28:
Try putting this in `/usr/lib/systemd/system-sleep/bat`:
```
#!/bin/sh
test "x$1" = "xpost" &&
  /usr/bin/echo 60 >/sys/class/power_supply/BAT1/charge_control_end_threshold
exit 0
```

HarshDobariya79, 2023-11-29:
> Try putting this in `/usr/lib/systemd/system-sleep/bat`:
> 
> ```
> #!/bin/sh
> test "x$1" = "xpost" &&
>   /usr/bin/echo 60 >/sys/class/power_supply/BAT1/charge_control_end_threshold
> exit 0
> ```
> 
> And make it executable: `sudo chmod +x /usr/lib/systemd/system-sleep/bat`

Worked! Thank you very much 😄

tshakalekholoane, 2023-12-01:
@pepa65 do you think this is a more robust way to set the threshold after sleep or is it enough of an edge case to leave things as is?

pepa65, 2023-12-01:
I have added it as an additional way to persist the threshold, just write that file on Persist and delete it on Reset.

tshakalekholoane, 2023-12-01:
I’ll mark this issue as resolved and if more people have the same problem maybe we’ll go with that too.

closed 2023-12-01 by tshakalekholoane