tshakalekholoane/bat #85: Will this work on NixOS ?

KingBain, 2024-02-28:
### Subject of the issue

Getting this error on NixOS when trying to make batter settings persist. 

Running the command:
**sudo /opt/asus-bat-manager/bin/bat persist --debug**

gives this output.
```
open /etc/systemd/system/bat-hibernate.service: read-only file system

goroutine 1 [running]:
runtime/debug.Stack()
	/nix/store/ws97rhhdzacbl58y8crpvg517zqnbsll-go-1.21.5/share/go/src/runtime/debug/stack.go:24 +0x5e
main.main.func1()
	/home/kingbain/bat/main_linux.go:156 +0x55
panic({0x52f700?, 0xc00011ca50?})
	/nix/store/ws97rhhdzacbl58y8crpvg517zqnbsll-go-1.21.5/share/go/src/runtime/panic.go:920 +0x270
main.main()
	/home/kingbain/bat/main_linux.go:222 +0x1230

```

Any chance you could create a nixos package ?

### Your system

For example,

OS: Manjaro Linux x86_64
Host: ZenBook UX425EA_UX425EA 1.0
Kernel: 5.13.8-1-MANJARO

### Version

0.14 (Latest)

tshakalekholoane, 2024-03-01:
> Any chance you could create a nixos package ?

I don't use NixOS so that could take a while. I am also not sure if that would work [based on a previous issue](https://github.com/tshakalekholoane/bat/issues/26#issuecomment-1152714589).

@pepa65 did you ever get it working on NixOS?

pepa65, 2024-03-01:
NixOS is a different kettle of fish that I have never fully entered. I'll leave that to people familiar with the inner workings of NixOS, as it doesn't want you plainly writing to its system files.