tshakalekholoane/bat #97: Error setting (end) threshold on Lenovo ThinkPad T14 Gen1 when the start threshold is higher

SylChamber, 2024-08-14:
Computer: Lenovo ThinkPad T14 Gen 1 AMD (manufactured in 2021)
CPU: AMD Ryzen PRO 7 4750U
OS: Ubuntu 24.04

I could not set a battery threshold on my Lenovo ThinkPad T14 Gen1 (while it works on my Asus ROG Strix G15 2021). I got the following error:

```shell
$ sudo bat threshold 80
A fatal error occurred. Please rerun the command with the `--debug` flag enabled
and file an issue with the resulting output at the following address
https://github.com/tshakalekholoane/bat/issues/new.
```

with the `debug` flag:

```shell
$ sudo bat threshold 80 --debug
write /sys/class/power_supply/BAT0/charge_control_end_threshold: invalid argument

goroutine 1 [running]:
runtime/debug.Stack()
	/opt/homebrew/Cellar/go/1.22.0/libexec/src/runtime/debug/stack.go:24 +0x5e
main.main.func1()
	/Users/tshaka/x/bat/main_linux.go:111 +0x57
panic({0x5383a0?, 0xc000124270?})
	/opt/homebrew/Cellar/go/1.22.0/libexec/src/runtime/panic.go:770 +0x132
main.main()
	/Users/tshaka/x/bat/main_linux.go:247 +0x1125
```

When searching on the battery system path, I found the [UbuntuHandbook page on setting battery threshold](https://ubuntuhandbook.org/index.php/2024/02/limit-battery-charge-ubuntu/). Trying to set the `end` threshold manually failed as well:

```shell
$ sudo sh -c "echo 80 > /sys/class/power_supply/BAT0/charge_control_end_threshold"
sh: 1: echo: echo: I/O error
```

It led me to try *manually* setting the **lower** charging threshold to a lower value than the `end` value:

```shell
sudo sh -c "echo 20 > /sys/class/power_supply/BAT0/charge_control_start_threshold"
```

Which I unfortunately did *before* trying to read the current value. :-( I don't know what it was before. Afterwards, I could set the `end` threshold, manually or with `bat`.

I gather that the `end` threshold must not be lower than the `start` threshold.

SylChamber, 2024-08-14:
Uh... actually, I think I got the threshold backwards. I'm comparing with values on my Asus ROG Strix G15. It only has **one** value: `charge_control_end_threshold`. And I set it to `80`. There is no `start` threshold. 

My Lenovo ThinkPad T14 G1 has two thresholds:

* `charge_control_end_threshold`
* `charge_control_start_threshold`

as well as two legacy APIs that are supposedly synchronized with those two.

pepa65, 2024-08-14:
Thank you for reporting this. Do you have any idea how `charge_control_start_threshold` should be handled? It seems like it could be set by the user, but maybe there is a sane default, depending on `charge_control_end_threshold`..?

SylChamber, 2024-08-14:
Yes, I found the way it works in the [Device Compatibility section on ThinkPad in the Battery Health Charging GNOME extension documentation](https://maniacx.github.io/Battery-Health-Charging/device-compatibility/thinkpad). It depends on the direction of the movement of the thresholds: up or down. **The threshold closest to the target must be modified first.** That's why `bat` failed to lower my ThinkPad's threshold from 100 to 80. It was decreasing the `end` threshold to a lower value than the `start`. And manually setting the `start` threshold to a lower value allowed `bat` to set the `end` threshold.

*Battery Health Charging* supports 3 profiles and their defaults define 5 percent windows with the sart/end thresholds:

* Maximum Lifespan:  `55-60`
* Balanced: `75-80`
* Full Capacity: `95-100`

If the thresholds are *decreased* (e.g. from `100` to `60`), then `charge_control_start_threshold` must be set first (e.g. to `55`), then `charge_control_end_threshold` (e.g. to `60`):

```shell
echo '55' | sudo tee /sys/class/power_supply/BAT0/charge_control_start_threshold
echo '60' | sudo tee /sys/class/power_supply/BAT0/charge_control_end_threshold
```

If the thresholds are *increased* (e.g. from `60` to `80`), then `charge_control_end_threshold` must be set first (e.g. to `80`), then `charge_control_start_threshold` (e.g. to `75`):

```shell
echo '80' | pkexec tee /sys/class/power_supply/BAT0/charge_control_end_threshold
echo '75' | pkexec tee /sys/class/power_supply/BAT0/charge_control_start_threshold
```

pepa65, 2024-08-14:
OK, I think just set the start value 5 lower than the end, and check which one needs to be set first.

tshakalekholoane, 2024-12-16:
I had a thought about how this could be implemented without breaking changes in the API. 

We could change the threshold argument to accept a range such that the user could pass a range to the `threshold` subcommand e.g., 

```shell
bat threshold 55,60
```

To maintain backwards compatibility, setting a single value would imply 0 as a lower bound, on devices that have a start threshold setting.

```shell
# The following are equivalent.
bat threshold 60
bat threshold 0, 60
```

What do you think?

SylChamber, 2024-12-17:
Personnally, I don't see the point of the start threshold either. I've been using the Battery Health Charging extension in GNOME for a few months and I never felt the need to customize the thresholds. I use the built-ins 60, 80 and 100, which set 55, 75 and 95 start thresholds on my ThinkPad T14. I would always set the start threshold 5 points below the end threshold.

I'm going to do some tests with that start threshold on my ThinkPad T14 because I don't understand what it does either.

By the way, I recently wrote over Ubuntu on that ThinkPad with another distro I wanted to try out, and I was suprised to see that the thresholds had persisted after the whole disk erasure. And I still could change them manually. Where could they be stored? In the TPM chip?

pepa65, 2024-12-17:
They get reset to 100/0 (no end threshold) when the module loads. When the module never loads again, the value must get stored somewhere in the UEFI/BIOS area.

pepa65, 2024-12-18:
OK, then `start` has to be set a little bit lower than `end`.

pepa65, 2024-12-18:
You can try the `batlimit` binary here to see if this works: https://github.com/pepa65/batlimit/releases/tag/0.6.0

SylChamber, 2024-12-19:
I noticed in the `Makefile` that you use `gofumpt` for formatting, and it requires go `1.22`, yet the `go.mod` file specifies version `1.21`.  Which version of go should I be using?

tshakalekholoane, 2024-12-19:
You can disregard the requirements of gofumpt or any other tool in the Makefile. For the application, any version >= 1.18 should be fine and you're welcome to bump it to a later version if you need to.