tshakalekholoane/bat #102: issue: Does not work with Linux Kernel >= 6.12

okarin69, 2024-11-26:
### Subject of the issue

With the arrival linux kernel 6.12, in my asus laptop the /sys/class/power_supply/BAT_0/charge_control_end_threshold file has disappered and charging threshold does not work. But when I switch to linux-lts kernel it reappers and the charging threshold works properly.

Hope this information is helpful.

### Your system

For example,

OS: Endeavour OS x86_64
Host: Asus Vivobook Pro 16 2022
Kernel: Linux 6.12.1

### Version

1.0 (Latest)

### Steps to reproduce

Switch to latest stable Kernel 6.12.1 or newer. =>  Error: Charge threshold file does not exist 

Going back to linux- 6.6.63-lts => Working properly

tshakalekholoane, 2024-11-27:
I cannot find anything to suggest this is a change in the kernel. The code that handles this hasn't changed for half a decade. Not withstanding any changes made upstream by your distribution.

Can you post the exact error you are getting from running the program? 

(Also, the threshold is at one of [`/ sys/class/power_supply/BAT{0,1,C,T}`](https://github.com/torvalds/linux/blob/aaf20f870da056752f6386693cc0d8e25421ef35/drivers/platform/x86/asus-wmi.c#L1440-L1443) not `/sys/class/power_supply/BAT_0/charge_control_end_threshold` so that could also be the source of your problem).

pepa65, 2024-11-27:
What is the output of `ls /sys/class/power_supply/` and `ls /sys/class/power_supply/BAT0` on the problematic kernel?

It could be that somehow this was not included in the built of that particular kernel, but the output would shed some light.

okarin69, 2024-11-27:
Output for  `ls /sys/class/power_supply/ `

`BAT0`

Output for ` ls /sys/class/power_supply/BAT0`

```
alarm           charge_full_design  device        power          subsystem   voltage_min_design
capacity        charge_now          hwmon1        present        technology  voltage_now
capacity_level  current_now         manufacturer  serial_number  type
charge_full     cycle_count         model_name    status         uevent
 
```

The charge_control_end_threshold file is present in  LTS kernel 6.6.63

The output when `sudo ./bat threshold 80` is run is `Charging threshold setting not found.`

pepa65, 2024-11-27:
I would make inquiries of the people who build the kernel. Unfortunately, there is nothing this project can do about it...

okarin69, 2024-11-27:
Yes I understand, but you guys are very well suited to know what's going on in the kernel or if there is a change in the process. Hope the word gets out and the problem is solved.

pepa65, 2024-11-28:
Do you have something like `/boot/config-6.12.1` on your system? You could check the output of `grep ASUS /boot/config-6.12.1` and `lsmod |grep asus`.

okarin69, 2024-11-29:
`grep ASUS /boot/config-6.12.1`
Output: `grep: /boot/config-6.12.1: No such file or directory`

`lsmod |grep asus`

Output:
```
asus_nb_wmi            28672  0
asus_wmi               98304  1 asus_nb_wmi
ledtrig_audio          12288  3 snd_ctl_led,snd_hda_codec_generic,asus_wmi
sparse_keymap          12288  1 asus_wmi
platform_profile       12288  1 asus_wmi
rfkill                 40960  8 asus_wmi,bluetooth,cfg80211
i8042                  57344  1 asus_wmi
video                  77824  3 asus_wmi,i915,nvidia_modeset
wmi                    45056  3 video,asus_wmi,wmi_bmof

```

pepa65, 2024-11-29:
At least we know the module is loaded..! I should maybe have asked for `ls /boot` first to see in a config is available there.

okarin69, 2024-11-29:
Here is the output for `ls /boot`

```
drwx------    - root  1 Jan  1970  efi
drwxr-xr-x    - root 25 Nov 05:59  grub
.rw------- 252M root 28 Nov 01:34  initramfs-linux-fallback.img
.rw------- 204M root 28 Nov 01:34  initramfs-linux-lts-fallback.img
.rw------- 131M root 28 Nov 01:34  initramfs-linux-lts.img
.rw------- 175M root 28 Nov 01:34  initramfs-linux.img
.rw-r--r-- 8.1M root 12 Nov 22:50  intel-ucode.img
.rw-r--r--  14M root 24 Nov 08:56  vmlinuz-linux
.rw-r--r--  13M root 25 Nov 04:15  vmlinuz-linux-lts

```

pepa65, 2024-11-30:
OK, so no config information in that directory. I think we would be looking for entries CONFIG_ASUS_NB_WMI (maybe CONFIG_ASUS_LAPTOP and CONFIG_SENSORS_ASUS_WMI).

okarin69, 2024-11-30:
How do I look up this information?
CONFIG_ASUS_NB_WMI (maybe CONFIG_ASUS_LAPTOP and CONFIG_SENSORS_ASUS_WMI).

pepa65, 2024-11-30:
Endeavour OS is Arch, right? Your kernel config might be in `/proc/config.gz`, try `ls /proc/config.gz` to see if it's there. if it is, you can do `zgrep ASUS /proc/config.gz` (if you have `zgrep`, which most likely you do).

okarin69, 2024-12-01:
Here's the output:

```
CONFIG_USB_PEGASUS=m
CONFIG_TABLET_USB_PEGASUS=m
CONFIG_SENSORS_ASUS_WMI=m
CONFIG_SENSORS_ASUS_EC=m
CONFIG_HID_ASUS=m
CONFIG_ASUS_LAPTOP=m
CONFIG_ASUS_WIRELESS=m
CONFIG_ASUS_WMI=m
CONFIG_ASUS_NB_WMI=m
CONFIG_ASUS_TF103C_DOCK=m
```

pepa65, 2024-12-01:
This looks like a similar problem: https://bugzilla.kernel.org/show_bug.cgi?id=219075

Biasio, 2024-12-01:
Same here, for me also bluetooth controller stopped working. I think it's something related to the last kernel

pepa65, 2024-12-02:
@Biasio What distro and kernel are you on?

pepa65, 2024-12-02:
The issue has already been reported and a fix been found:
https://bugzilla.kernel.org/show_bug.cgi?id=219517

okarin69, 2024-12-04:
@pepa65 Does this fix reflect in the kernel in later stages? In linux-6.12.1 I am still facing the issue.

okarin69, 2024-12-08:
Since the program is working correctly and the problem is with the kernel, I am closing this issue.

What I have found is that the asus_nb_wmi driver is not working properly. Hope we get a fix soon. Until then LTS kernel it is.

tshakalekholoane, 2024-12-17:
Let's keep it open for now to avoid duplicate bug reports (see #106).