tshakalekholoane/bat #104: issue: Charging threshold setting not found.

eugenialoli, 2024-12-09:
### Subject of the issue

I'm using a Macbook Air 2015, and it says that "Charging threshold setting not found". Going inside the folder BAT0 it has only these inside: 
alarm         current_avg  manufacturer  subsystem   voltage_min_design
capacity            current_now  model_name    technology  voltage_now
charge_full         cycle_count  power         temp        wakeup42
charge_full_design  device       present       type
charge_now          hwmon2       status        uevent

Any chance it has that feature, to charge it up to a point only but it's named differently?

### Your system

Macbook Air 2015
Mint 22
6.8 kernel

### Version

1.2 (Latest)

tshakalekholoane, 2024-12-09:
It seems unlikely. Based on the contents of your BAT0 folder, those variables all have different purposes unrelated to setting the charging threshold.

eugenialoli, 2024-12-09:
If I change the battery with a modern, third party one could there be a chance of working? I guess what I ask is, is the variables reported directly from the battery, or the BIOS?

pepa65, 2024-12-10:
Interesting question! It would need to be an ASUS battery, but I think it's not the battery, it's the charger that is part of the board that has the capacity to expose this through the kernel module `asus_nb_wmi`. I think if you load it, you will likely get the report back that it cannot load the module, because there is "No such device".

By the way, so fun to see you here, I was an avid OSNews reader decades ago..!

tshakalekholoane, 2024-12-10:
I don't think it's reported from the battery too. My estimation is that that you'd need a device that is from one of the manufacturers that have drivers in the kernel for this which so far seems to be [ASUS](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/asus-wmi.c), [Lenovo](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/thinkpad_acpi.c) (ThinkPads), and [Dell](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/dell/dell-laptop.c).

SylChamber, 2024-12-11:
I suggest having a look at the extensive documentation for the Battery Health Charging extension for GNOME. They [support and have a page for Intel Macs](https://maniacx.github.io/Battery-Health-Charging/device-compatibility/apple-intel-series). Maybe that all that is required is installing the [applesmc-next](https://github.com/c---/applesmc-next) kernel module like they write.