tshakalekholoane/bat #106: issue: Charging threshold setting not found. [Zenbook UM425QA]

Rekt3y, 2024-12-16:
### Subject of the issue

After a Fedora 41 update, charging control no longer works with the message: Charging threshold setting not found.

### Your system

OS: Fedora KDE 41
Host: ZenBook UX425QA_UM425QA
Kernel: Linux 6.12.4-200.fc41.x86_64

### Version

1.2 (Latest)

### Steps to reproduce

1. Install Fedora 41, and update all packages.
2. Install bat
3. Try setting threshold to 80

### Expected behaviour

It should set a charge limit.

### Actual behaviour

It fails with the not found message.

Rekt3y, 2024-12-16:
I booted the previous kernel I had, and charge control still works there. That kernel is Linux 6.11.10-300.fc41.x86_64
Hope this helps

tshakalekholoane, 2024-12-17:
This seems to be a duplicate of https://github.com/tshakalekholoane/bat/issues/102 caused by a regression in the kernel.

Rekt3y, 2024-12-26:
Kernel 6.12.5 on Fedora 41 has fixed this issue.

closed 2024-12-26 by Rekt3y