tshakalekholoane/bat #117: Laptop charging past the set threshold

sergemuur, 2025-08-21:
Hello there, @pepa65 and @tshakalekholoane ! Sorry for reposting my question here (from the [discussion section](https://github.com/tshakalekholoane/bat/discussions/73#discussioncomment-14134915)), I hope you don't mind🙏.

I've been happily using bat since you guys helped me the last time few years ago, but it seems I messed it up once more after a clean arch installation on the same machine (finally ditched Windows entirely).

_ASUS UM3402YA
archlinux 6.16.1-zen1-1-zen_

This time I tried downloading the binary from your releases page and using `chmod` and, probably, my stupid imagination.
These are the exact commands I ran:
```
sudo chmod +x /home/sergemuur/Programs/bat
sudo install /home/sergemuur/Programs/bat /usr/local/bin/
```
The program was installed and nothing seemed off (I set the threshold and ran `persist`), except the laptop kept charging past the set threshold. I did the `sudo bat reset` and deleted the initial folder in `/home`, also looked for the services you had pointed out the last time in [the discussion](https://github.com/tshakalekholoane/bat/discussions/73), but none of the services was found.

I reinstalled bat multiple times using good old `git clone` and `make build`, but it never worked (kept charging past the threshold I set). 
The /`etc/systemd/system/bat@.service` file was properly deleted by the reset command each time.

The `bat-multi-user.service` is active and enabled:
```
● bat-multi-user.service - Persist the battery charging threshold after multi-user
     Loaded: loaded (/etc/systemd/system/bat-multi-user.service; enabled; preset: disabled)
     Active: active (exited) since Sun 2025-08-17 22:26:18 MSK; 3h 23min ago
 Invocation: 7e5c3f1b75c24870ab230bb48403ab11
   Main PID: 978 (code=exited, status=0/SUCCESS)
   Mem peak: 1.7M
        CPU: 9ms

Aug 17 22:26:18 archlinux systemd[1]: Starting Persist the battery charging threshold after multi-user...
Aug 17 22:26:18 archlinux systemd[1]: Finished Persist the battery charging threshold after multi-user.
```
The `/sys/class/power_supply/BATT/charge_control_end_threshold` file has the value of the threshold I set inside.

The only strangeness is that I haven't been able to find any other services running (like `bat@suspend.service` or `...hibernate` etc.) except for the `bat-multi-user.service`.
```
[sergemuur@archlinux ~]$ sudo systemctl status bat@suspend.service
Unit bat@suspend.service could not be found.
```
Maybe it could be connected with the fact that my system is on btrfs now? I heard that it has a hard time handling hibernation and stuff.

It seems that the answer is very simple but I've run out of any ideas as to how to find it myself. Please, I would very much appreciate your help yet one more time with this!

P.S. Oh, one more thing. Every time I reinstall bat, it somehow remembers the threshold I set during previous installation. By default it's not 100. Idk why the `reset` fails to reset it.

pepa65, 2025-08-22:
When the `/sys/class/power_supply/BAT?/charge_control_end_threshold` is set, it should work. But right now, even though this setting is exposed, and I have it set, the driver seems to ignore it and charge past the set limit here... I am not sure what is happening, but it is not at the level of this tool, but at the level of the kernel driver. This tool just sets and optionally persists the setting through systemd services.

pepa65, 2025-08-22:
I just installed a 6.14 kernel it but still keeps charging past the value set in `/sys/class/power_supply/BAT?/charge_control_end_threshold`...

kevlar700, 2026-09-02:
I can confirm that charge_control_end_threshold is ineffective on Debian Trixie too. It used to work fine but no longer does with recent kernels (6.12.101)!