tshakalekholoane/bat #118: minor bug: battary health over 100%

CharaVerKys, 2025-08-31:
my battery have energy design > than actual full energy
just clamp value to 100, where health: goto's label

// also i personally not like one char var names, but who care

CharaVerKys, 2025-09-02:
> I am not sure clamping the value is the answer if our assumptions are wrong to begin with.  

my battery design is 29, actual is 30
info from upower
there literally nothing to be wrong to begin with

wait....
that actual error with print `>` instead of `<`
idk how that happened, and i didnt re-read what i wrote

*i not mind to screenshot, i even deleted packet after learning how units works, i was on avoiding systemd configs, cuz never required to actually deal with units, and always wrote my script in bashrc,
but now im actually need to run command before login, on root lvl*

```go
var health_ = x * 100 / y 
if(health_ > 100){health_ = 100;} // actually possible in real world
fmt.Println(health_)
```

pepa65, 2025-09-02:
So you're saying that `/sys/class/power_supply/BAT?/charge_full_design` is smaller than `/sys/class/power_supply/BAT1/charge_full`?

That would mean that the driver is very broken...

tshakalekholoane, 2025-09-03:
Let’s go with that then. 

Thank you for filing the issue @CharaVerKys. Feel free to submit the fix if you’re open to that.